rudetrooper/Octoprint-Chituboard #1: Support for encrypted printers (Mars 3, etc..) without the SDK

PaulGameDev, 2021-08-23:
Hey there!

CryptoCTB have yesterday, after awaiting release of the SDK, released [LiberatingMarsCLI](https://github.com/CryptoCTB/LiberatingMarsCLI) as a free, open-source counter-response, licensed under the MIT License.

I'd love it if you take a look at it to understand the new file format and understand how to decrypt the files for reading!
For more info, feel free to contact me on Discord! Paul_GD#4195

rudetrooper, 2021-08-23:
Thanks I'll look into your code, if it's just an AES encryption it should be easy enough to figure out. Is it okay if I reimplement some of your code in python for plugin compatibility reasons?

PaulGameDev, 2021-08-23:
> Thanks I'll look into your code, if it's just an AES encryption it should be easy enough to figure out. Is it okay if I reimplement some of your code in python for plugin compatibility reasons?

Of course! Feel free to use it however you find it fitting! :D

altaic, 2021-10-01:
@PaulGameDev nice work! Would I be right in thinking that this should work for any Chitu-based MSLA printer (e.g. Anycubic Photon Mono X) with recent firmware?

jremen, 2022-08-17:
@rudetrooper 
Any updates on this? Mariner seems to be dead, no updates, this plugin also seems not to support any new printers due to failing to implement CTB encrypted files support…

TRusselo, 2022-12-27:
Would love to see support
I dont see anyone has posted the "open source" files provided by elegoo for the Mars3 4k mono
https://github.com/elegooofficial/Mars3

this might put some movement on the project

PaulGameDev, 2023-01-05:
> Would love to see support I dont see anyone has mention the open source files provided by elegoo for the Mars3 4k mono
> 
> Aug 12 2021 - statement https://www.elegoo.com/blogs/news/newest-statement-of-the-chitubox-board-firmware-on-mars-3 Oct 09, 2021 - release statement https://www.elegoo.com/blogs/news/open-source-files-for-elegoo-mars-3-ultra-4k-mono-lcd-3d-printer https://github.com/elegooofficial/Mars3
> 
> this might put some movement on the project, i hope

The "open-source" release of the Mars 3 has limited itself to the physical CAD files, leaving out the electronics and firmware completely. So sadly it has basically zero impact on this, besides maybe making it easier to retrofit an open resin controller board (PrometheusMSLA Apollo for example).

rudetrooper, 2023-01-05:
Basically what PaulGameDev said. I'm working on it, but I haven't had time to finish adding the support for the files into the plugin. The code is mostly done, it will probably be finished sometime soon. Just need time to test it.