rudetrooper/Octoprint-Chituboard #2: Long delay sending Z changes? Also camera freezing?

Arakon, 2021-09-03:
Hi.
I'm running a Pi 3A+ and a Mars Pro. I switched from Mariner on a Zero to OctoPi just today.
For printing, it seems to work great.
However, when I send a Z change (i.e. Home, or move up/down), it takes up to 30 seconds before the actual command is executed. The terminal is stuck at "Send: M4000" during that time.
Also, the current layer update is sometimes missed completely, so the layer number stays at i.e. 8 for 8 seconds, then suddenly jumps to 11 or 12.

If you need any other details, let me know please.

rudetrooper, 2021-09-03:
The delay on the Z changes is due to the code updating the when the file percentage increments by 1%. Its an inaccurate method I'm going to fix it once I have some time next week. The Z change issue is new to me, I think its due to the printer firmware slow communication rate. It takes some time to respond to Z change and home commands because it can only respond to 1 command per 0.5 seconds and the HAL they used is probably shitty. Or my shitty code is blocking the send loop. What printer are you using?

Arakon, 2021-09-03:
Mars Pro with the original Chitu Board (V2, I guess?).

rudetrooper, 2021-09-03:
Can you send me a picture of how you hooked it up. I need to create a wiki with hardware setup pictures eventually. Also I did some research about your camera problem and it might be an issue with the pi 3A+. Unfortunately I don't have one to test right now.

Arakon, 2021-09-03:
I can take a picture sometime tomorrow, although it's already all closed up now.
I just experimented a bit more.

- First time I send a Z command (i.e. up 1mm), the last thing on the terminal is "Send: M4000" for anywhere between 10 and 30 seconds. Then the axis moves.
- If I am on the camera view, the camera freezes as soon as the Z-axis starts moving.
- If I click away from the camera tab i.e. to Terminal and back, the camera resumes working. Move the Z-Axis, it freezes again, but only once Z moves, which can still take 5+ seconds.

It seems to me that sending the Z command causes something in the system to drop the camera connection. Note that this does NOT happen during regular printing (which has a ton of Z changes), the camera remains active the whole time then.

Also, I just noticed the following in terminal when switching on the printer:

```
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyS0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x69f05bb0, open=True>(port='/dev/ttyS0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: M4002
Recv: ok V4.3.13_LCDC
Repl: ok V4.3.13_LCDC
 -> ok start
Send: M4002
Recv: ok V4.3.13_LCDC
Repl: ok V4.3.13_LCDC
 -> ok start
Send: M4002
Changing monitoring state from "Connecting" to "Operational"
Recv: ok V4.3.13_LCDC
Repl: ok V4.3.13_LCDC
 -> ok start
Send: N0 M110 N0*125
**Recv: //####checkSum error,i:2 expect:0x7e,actual:0x7d!str:N0*125**
Recv: 
Recv: ok
Send: M115
Recv: ok CBD make it.Date:Sep  9 2020 Time:10:56:30
Repl: ok CBD make it.Date:Sep  9 2020 Time:10:56:30

 -> ok FIRMWARE_NAME:CBD made it PROTOCOL_VERSION:V4.3.13_LCDC
 Date:Sep  9 2020 Time:10:56:30

```

rudetrooper, 2021-09-03:
The M110 is due to an issue with how the firmware handles the M110 set line number command. You should add M110 to the blocked commands list and change "Send a checksum with the command" to never. I'll modify my plugin so that's done by default. During regular printing all the Z changes are handled on the printer side. The slow serial communication is a major issue, that's why I set the polling intervals to 2s.

Arakon, 2021-09-03:
Okay, that took care of the checksum display, thanks.

Arakon, 2021-09-04:
[image]
[image]

As promised, pictures of the setup. Note that this is temporary, I'm waiting for a much more compact USB cable still, and the case was just attached with a small piece of double sided tape to get it up and running.
The camera doesn't deliver any sort of great image from inside the build chamber (mostly red), but it's more of a safety thing.

Octoprint is also running the Tasmota plugin and a tasmota-powered socket, so that I can turn the printer on and off from within Octoprint (the Pi is always on, running on a seperate power supply).

rudetrooper, 2021-09-04:
Thats better looking than my janky setup. The chituboard firmware doesn't support checksums, its a feature in Marlin firmware.

Arakon, 2021-09-15:
[image]
[image]

Updated with the finalized build pictures. 
I also verified that the camera freezing is caused by the plugin.. I tried 2 different cameras, 2 different ribbon cables.
As soon as I disable the plugin and reboot, the camera stays active all day long just fine. Enable the plugin, and it will freeze randomly.

xmodpt, 2021-10-20:
i am having exectly the same issues with the delay.... it takes um to 5min to get a comand sent and executed.
On my terminal all i have is this :

Send: G91
Recv: wait
Repl: wait -> echo:busy processing
Recv: wait
Repl: wait -> echo:busy processing
Recv: wait
Repl: wait -> echo:busy processing
Recv: wait
Repl: wait -> echo:busy processing
Recv: wait
Repl: wait -> echo:busy processing
Recv: wait
Repl: wait -> echo:busy processing
Recv: wait
Repl: wait -> echo:busy processing
Recv: wait
Repl: wait -> echo:busy processing
Recv: wait
Repl: wait -> echo:busy processing
Recv: wait
Repl: wait -> echo:busy processing
Recv: wait
Repl: wait -> echo:busy processing
etc.......

any ideas ?

Thank you 
NP