rudetrooper/Octoprint-Chituboard #3: Timelapse not stopping

Arakon, 2021-09-15:
Not entirely sure if this is actually related to the Plugin itself.

When running a timed timelapse while printing, the timelapse starts automatically when the print starts, but will not finish when the print stops.. it just continues recording forever until manually stopping it.
I don't think this happens when printing from SD on a FDM printer, so I'm thinking some sort of "Finished" signal to Octoprint is missing?

rudetrooper, 2021-09-16:
I'll look into it this weekend. It could be due to some error in how I'm handling the gcode receive loop before its passed into the comm layer.

zeroecho, 2022-04-13:
Did you ever get this resolved? Noticed the same thing when I tried one today

rudetrooper, 2022-04-13:
No, it's been hard to track down the cause of the bug.