rudetrooper/Octoprint-Chituboard #4: Will this support Anycubic Photon Mono X printers?

DarumasLegs, 2021-09-16:


rudetrooper, 2021-09-17:
It should work if you connect the RX and TX pins to the serial port used by the WIFI chip. I don't own an Anycubic photon Mono X or have the mainboard pinout so I don't know if there's another set of accessible serial pins. If you get it working send me some pictures of the wiring to the mainboard so I can add it to the README.md.

etyrnal, 2021-12-01:
Would some or all of this apply to the Photon Mono SE?  I just ordered one, and then after the unit left the factory i discovered that the built in wifi functionality is super lame and was hoping i could add it to my octoprint culture.

rudetrooper, 2021-12-02:
The Anycubic Photon mono SE should also work if you disconnect the USR-C322 chip being used for wifi and connect the RX and TX pins to the serial port used by the WIFI chip. I already added in support for the file type used by the Photon Mono SE. If you get it working can you post a picture of the wiring setup and I'll add it to the supported printers list.

rudetrooper, 2021-12-02:
> ## Problem
> I had a look into my printer. Unfortunately there's no mirrored UART port.
> 
> ## Solutions
> ### UART
> So you basically have 2 options for using UART:
> 
> 1. drop the native (feature less) WiFi support (I have not tried out, if the printer will work without wifi module)
> 2. split the UART using an IC (you can't use an electrical Y here) which replicates the messages received to two ports and makes sure only one input can write.
> 
> ### WiFi
> There is, however another option: Use the service exposed on port 6000 by the WiFi-module.
> 
> #### Commands
> I have intercepted the communication of the mobile app and the printer and identified the following commands
> 
> ##### General
> * Requests are terminated by `,`
> * (most) Responses are terminated `,end`
> 
> ##### Get the information about the printer
> `sysinfo,`
> 
> > `sysinfo,<PrinterType>,<FirmwareVersion>,<SSID>,end`
> 
> ##### List all files
> `getfile,`
> 
> > `getfile,<files>,end`
> 
> Whereas files is a comma separated array of `<filename>/<alias>` (e.g. `baby-yoda.pwmx/0.pwmx,super-trooper.pwmx/1.pwmx`). The alias must be used in the requests (probably this is a way to reduce communication overhead).
> 
> ##### Get the preview of a file
> `getPreview2,<alias>,`
> 
> > `<binaryData>`
> 
> I need to hexdump this and analyze it, the image will probably be PNG or JPG (for simplicity) and the response most likely will contain a header which denotes the size of payload (image).
> 
> ##### Start a print
> `goprint,<alias>,`
> 
> > `goprint,OK,end`
> 
> ##### Stop a print
> `gostop,`
> 
> > `gostop,OK,end`
> 
> ##### Get the status of the printer
> `getstatus,`
> 
> > `getstatus,<status>,end`
> 
> Whereas `<status>` is:
> 
> * `stop`: printer is stoped
> * `print,<filename>/<alias>,<layerCount>,<number(probably cur layer)>,<number>,<number(probably cur time)>,<printTime in sec>,<requiredResin>,UV,<number>,<layerThinkness>,<number>,end`
>   Example: `getstatus,print,baby-yoda.pwmx/0.pwmx,1548,0,0,0,26373,~139.25mL,UV,139.25,0.05,0,end`
> 
> Note: I need to verify some of the values, though.
> 
> ##### Whatever mode is (probably alive check)
> `getmode,`
> 
> > `getmode,0,end`
> 
> ## Suggestion
> Instead of putting a cable to the port, why not using a TCP connection and send the commands and receive the status from the service? - This would save all Photon Mono X users opening up the printer :)
> 
> ## Contribution
> I have not looked into the source code of octoprint (not a huge fan of python, sorry). But IMHO this could be doable (e.g. using socat and a virtual TTY), but I would appreciate some help.
> 
> Would you be interested in that?
> 
> regards, christian

At the moment Octoprint only supports the UART solution. The TCP solution would require a significant refactor of Octoprints backend communication code since at the moment it basically acts a gcode sender over a serial connection. Octoprint 2.0 will support wifi connections once Foosel is done refactoring the comm layer. Unfortunately I don't have time at the moment to re-engineer octoprint to support a single printer version.

bwright86, 2022-03-19:
I'm currently taking the plunge to get an Anycubic Photon Mono X hooked up with Octoprint and this plugin.

Here are some pics of the mainboard, the wifi board (USR-C322), and the cable connections.

I'm still working to get it connected via Serial link. I'm using a pi 2 at the moment, so no OTG or wifi... darn chip shortage!
[image]
[image]
[image]

bwright86, 2022-03-22:
So, I will go ahead and say, that just removing the Wi-Fi modules isn’t enough to integrate. I haven’t been able to inspect the serial traffic, but it isn’t expecting gcode to connect and communicate.

The printer does function without the Wi-Fi module, but the protocol is likely non-standard.

I need to find a good way to hexdump the output. But you make a good point about it being transparent communications, meaning I could sniff the tcp packets for data.

As an aside. It does appear a 2nd UART port is wired to the blank pins of the Wi-Fi module, based on the main board chip’s data sheet, this is UART5. It doesn’t appear to be used, but if Anycubic team is interested in integrating for OctoPrint, either UART link could be utilized.

jordiMartinB, 2022-03-22:
The road has already been paved as there is  a library in development with all the commands exposed. Check https://github.com/adamoutler/anycubic-python for more info
As it was pointed the TCP solution would require some serious plumbing to make it work.

It would be interesting to check the 2nd UART port for life signals and see if it can be used instead. 
@bwright86  where did you find the info on the mainboard and do you know which pin has the TX and the RX for the 2nd UART ?

Regards,

Jordi

bwright86, 2022-03-23:
@jordiMartinB that project looks fantastic! and appears to be a good interim solution for this printer.

I was just starting to accept that I could only do remote file-management, but full remote printer management wasn't quite feasible yet.

I stumbled across UART5 from the datasheets here:
https://www.st.com/en/microcontrollers-microprocessors/stm32f407ze.html

It appears this is the LQFP144 package, with the pinout diagram on pg. 43, and `UART5` being pins 113 and 116. Which are curiously broken out to the blank pins for the wi-fi board. It appears `UART4` are the pins used by the wi-fi board.

From my attempts, using Octoprint's "Auto" baudrate, i didn't see any signs of life on `UART5`. Nor did i see any life when i power cycled the printer.
`UART4` does show some life when power cycling the printer, and the output appears to be an initial connection from the mainboard. Likely to wake up the wi-fi board and signal it to connect to an Access Point. I haven't been able to hexdump the output and convert to AT commands, but I am curious.

For the USR-C322 Wi-fi module, the datasheet and other documents can be pulled from here:
https://www.pusr.com/products/ti-cc3200-wifi-module.html

Thanks again,
Brent

etyrnal, 2022-03-23:
are there any similarities between the mono x and the mono SE where both
machines will benefit from figuring out one or the other? or are they both
unique beasts?

jordiMartinB, 2022-03-24:
Hello,

After peeping the firmware with a HexEditor it looks like the photon SE exposes the same commands.

The anycubic-photon project lists the following 

 Command: getstatus - Returns a list of printer statuses.
    Command: getfile - returns a list of files in format <internal name>: <file name>.
        When referring to the file via command, use the <internal name>.
    Command: sysinfo - returns Model, Firmware version, Serial Number, and wifi network.
    Command: getwifi - displays the current wifi network name.
    Command: gopause - pauses the current print.
    Command: goresume - ends the current print.
    Command: gostop,end - stops the current print.
    Command: delfile,<internal name>,end - deletes a file.
    command: gethistory,end - gets the history and print settings of previous prints.
    Command: delhistory,end - deletes printing history.
    Command: goprint,<internal name>,end - Starts a print of the requested file
    Command: getPreview1,<internal name>,end - returns a list of dimensions used for the print.
    Command: getPreview2,<internal name>,end
    - returns a binary preview image of the print.
    Those are also listed and found in the firmware but no confirmation on their inner workings and function
    Command: detect - unknown
    Command: stopUV - unknown
    Command: getpara - unknown
    Command: setPara - unkwown
    Command: getmode - unknown
    Command: setname - unknown
    Command: getname - unknown
    Command: setwifi - unknown
    Command: setZero - unknown
    Command: setZhome - unknown
    Command: setZmove - unknown
    Command: setZstop - unknown

It looks like from both files that the UART rate is set to 460800.

Regards,

Jordi

rudetrooper, 2022-03-24:
> So, I will go ahead and say, that just removing the Wi-Fi modules isn’t enough to integrate. I haven’t been able to inspect the serial traffic, but it isn’t expecting gcode to connect and [...]

I've contacted Anycubic about this in the past and they aren't really interested in integrating with Octoprint.d

gfuchs3, 2023-09-08:
It would be good to sniff the session from Chitubox to the Mono X during a network send. I couldn't get it to detect my printer, but once I do - it should reveal the command sequence it uses for sending files.

gfuchs3, 2023-09-12:
detect command seems to be doing what the tools/detection menu item is doing - briefly turning on the UV with a rectangle frame light. returns OK (tested on mono x).

gfuchs3, 2023-09-12:
the getname command initially returned some non-text chars followed by the SSID. After sending "setname,<name>,end"
it seems to retain that name, including after a power cycle of the printer, and it is returned in commands "getname,end". That could be used to identify multiple printers on the network (?).

gfuchs3, 2023-09-12:
Not a great success with setwifi - I have another SSID with the same password as the one the printer was connected to originally. After sending "setwifi,"new SSID",end" the wifi SSID configured in the printer changed (notice "new SSID" is the alternate SSID I was trying to configure in the printer). However, following a power cycle - the printer was not able to get an IP address. I had to reset the wifi information (printing the Wifi.txt file) to get it restored. My guess is that it expects the Wifi password as an additional parameter of the command.

Hermes761, 2025-03-01:
Good morning.
Can I also use this wifi module (USR-C322) for my Photon Mono 4K? I would need a datasheet of my printer board. I don't think it is enough to just get the module and connect it to the connector. What is improved with this modification? It would be impostant and very interesting to have a much better performing Bluetooth module in this case. I have problems with the usb port (which I will check for 5v today) and I hope to solve all these inconveniences with this wifi module. Thank you very much.