rudetrooper/Octoprint-Chituboard #5: Support for Voxelab SLA Printers ? (.fdg)

doucettom, 2021-10-19:
I own a pair of Voxelab Proxima's that I'd like to use with your plugin. They're actually supported by mariner3D so they should be "supportable" by the plugin (no Chitu encrypted stuff)

I'm willing to test stuff as needed

TIA

rudetrooper, 2021-10-19:
FDG files will work with the plugin once I finish adding the read_dict method to fdg.py and add it to the allowed extension list in settings. I've just been busy with other stuff over the last 2 months. FDG file format support will be added this week.

doucettom, 2021-10-19:
Awesome ! I'll test as soon as you are ready. Is the webcam tab still 
available ? I have pi cams on mines. It's a matter of swapping SD cards to 
test :)

closed 2021-12-15 by rudetrooper

doucettom, 2021-12-19:
Just done a fresh install on my pre-bought SD card. Everything works up to the .fdg file upload. I didn't print anything yet but I should do some printing during the Holidays. Is 3D file thumbnail supposed to be shown ?

rudetrooper, 2021-12-20:
Not really, I never got around to properly integrating the file preview since I don't really use it. I'll try to find time to finish the layer preview stuff but I can't really make any promises about timelines.