rudetrooper/Octoprint-Chituboard #7: Support Mars printers with newest firmware

DivineJimmi, 2021-11-06:
Hey, my Mars 2 Pro is already prepared for Octoprint with your plugin. But it also has the newest firmware. I am now wondering, is it possible for me to use your plugin or will it not work? Your page only says "maybe" it won't work. I don't want to waste time trying it out just to find out, that its not working.

Sorry for my bad english.

Greetings from Germany

rudetrooper, 2021-11-10:
The unfortunate reason it doesn't work with the 1.9 firmware is because Chitu systems added some badly done file encryption to their .ctb files. I'm not going to add their closed source SDK into a FOSS plugin to support the new encrypted filetype. The other option is breaking the encryption (already done) but I don't want to cause any legal problems for Foosel(Main Octoprint dev). Your only option to use the plugin is to downgrade your printer firmware to the previous version.

DivineJimmi, 2021-11-10:
Thank you for your answer. Is there somewhere a tutorial hos to downgrade the printer?

closed 2021-12-02 by rudetrooper