rudetrooper/Octoprint-Chituboard #8: Support for Photon S?

t3hchad, 2021-12-01:
Tried to install on Photon S.  Gets connection error message.   Will this printer be supported at anytime?

Thank you

rudetrooper, 2021-12-01:
The Anycubic Photon S filetype is supported and it should work if you've properly setup the serial connection. The printer isn't listed under supported printers because I don't have access to one to verify it works. Can you share the error log and a picture of how you set it up(wires from pi GPIO serial -> printer mainboard)? Also did you run the Chituboard.sh script?

etyrnal, 2021-12-01:
following.

t3hchad, 2021-12-05:
I have 3 printers already setup.  2x Creality LD-002r and 1x Anycubic Photon.  Both are working flawlessly.  I set the Anycubic Photon S the same as the other 3.  See pics/logs below.
[image]
[image]
[octoprint.log](https://github.com/rudetrooper/Octoprint-Chituboard/files/7656374/octoprint.log)

rudetrooper, 2021-12-06:
Did you try using the RX2, TX2 pins?

t3hchad, 2021-12-06:
Yes.  I get the same results on the RX2, TX2.

t3hchad, 2021-12-06:
Also there is interference with having the USB cable plugged in.  The cable does have the power pin taped off.  Swapped know cables on other printers to this one.  Same effect.  The LCD freezes up on first boot if USB cable is plugged in.

etyrnal, 2021-12-06:
I'm waiting on my Photon Mono SE to show up in the email.  When it arrives, it's my intention to get this setup working with a RPI Zero 2W.  

This is the photo on the anycubic website for a replacement mobo.  It appears to show the GND RX and TX pins.  Those would be the pads i connect to the RPI Zero 2W, correct?

[image]

rudetrooper, 2021-12-06:
Those pins are probably for connecting an ESP-01S for WiFi. You should be able to use those pins. It might also work for the Anycubic Photon S. I suggest asking on the Anycubic Photon Printer Owners facebook group, some Anycubic staff hang out there and might be able to tell you if those serial pins are usable.

MSpenn01, 2022-02-15:
following any updates?

Padi1111, 2022-12-07:
I'm having exactly the same problem @t3hchad, did you find anything ?

t3hchad, 2022-12-07:
> I'm having exactly the same problem @t3hchad, did you find anything ?

Never found an issue.  Used the PI with octoprint for wireless file trasfer.  Wasnt able to get the PI to control the printer.

SithLordSid, 2024-01-02:
> I'm waiting on my Photon Mono SE to show up in the email. When it arrives, it's my intention to get this setup working with a RPI Zero 2W.
>
> This is the photo on the anycubic website for a [...]

My device was configured exactly as yours but I could not get it to communicate.