rudetrooper/Octoprint-Chituboard #10: Scarcity of raspberry pi zero 2W

etyrnal, 2021-12-29:
In light of the fact that raspberry pi zero 2W is so scarce in the usa right now, will it be advisable to just go with a Raspberry PI 4B?  Or is there some reasons the 4B wont' work, or requires difficult, additional, or undeveloped steps?  

If i can manage to finally source a raspberry pi zero 2w for my **Photon Mono SE**, will the **zero 2w** be just barely enough?  Or more than sufficient?  If it's literally all i would ever need, i'd prefer to just get a zero 2w because it's smaller, and cheaper, but i don't want to run into a situation where i wished i'd gone with something more capable.

If the zero 2w is just plain going to be 'impossible' to source, i would grab a **4B**.  Then, in the case i couldn't get it going with the **Photon Mono SE**, at least i could use the 4B for another project.

Nice thing about the zero 2W is that it's so small i can just velcro it on the side/back of the machine.

Anyway.  **Main question is, whether it's just as easy to set up with a 4B -- or is that road not paved yet?**

rudetrooper, 2021-12-29:
You can use the Raspberry pi 4B if you want. Thats what I used when I developed the plugin. I just recommended the Raspberry Pi Zero 2W because I personally like the form factor and its potentially cheaper than the Raspberry Pi 4B.

etyrnal, 2021-12-29:
Thanks!  Can the current version of the zero 2W handle all current functionality, and intended future plugin functionality without being in danger of choking?  If it can, i'd like to go that route.  Feels crazy paying $50 for a $15 board.  That was an additional reason behind considering getting a 4B instead.   Thanks again for your insight!

Has anyone tested this with the Photon Mono SE?

rudetrooper, 2022-01-06:
The Zero 2W can handle everything, I use it to run Octoprint for my Elegoo Mars 2. I think some people have tested it with the Anycubic Mono SE but they never told me if they got it working. I assume they did since they probably would have asked me for help if it wasn't working.