rudetrooper/Octoprint-Chituboard #13: Disk init fail

elhajj33, 2022-01-19:
Printer is a flashforge foto 8.9
Rpi is a Zero 2w

I can communicate with the printer over serials and I can see the files in the resin directory.  See below:
/dev/loop0      4.0G   16M  4.0G   1% /home/pi/.octoprint/uploads/resin

However, i get his error whenever i try to print:

Send: M23 Xp_2.2s.fdg
Recv: //Disk init fail!
Recv: //Disk init fail!
Recv: //############sd_mounted Error! cann't open file !
Recv: ok N:964
Send: M4000
Recv: ok B:163/0 E1:159/0 E2:158/0 X:0.000 Y:0.000 Z:0.000 F:0/256 D:0/0/0 T:0
Repl: ok T:0 /0 B:163 /0
 E1:159/0 E2:158/0 X:0.000 Y:0.000 Z:0.000 F:0/256 D:0/0/0 T:0

I'm not sure what the issue is.  I see the files in octoprint and in the directory when I ssh....

rudetrooper, 2022-02-05:
The printer isn't mounting the USB mass storage device for some reason. Did you properly connect the micro USB cable to from the USB-OTG port on the pi to the printers USB port?

zeroecho, 2022-04-11:
Is USB-data cable likely the cause for all disk init errors @rudetrooper ? Tried 2, one with 5v taped and another with red wire cut

Here's what I see in terminal after connecting:

Repl: ok CBD make it.Date:Sep  9 2020 Time:10:56:30
 -> ok FIRMWARE_NAME:CBD made it PROTOCOL_VERSION:V4.3.13_LCDC
 Date:Sep  9 2020 Time:10:56:30

Send: M21
Recv: //Disk init fail!
Recv: ok N:2
Send: M20
Recv: Begin file list
Recv: //Disk init fail!
Recv: //Disk init fail!
Recv: End file list
Recv: wait
Repl: wait -> echo:busy processing
Recv: ok L:0
Send: M4000

crimsonclyde, 2023-01-01:
Hi,

I end up with nearly the same, or exactly the same, what @elhajj33 and @zeroech posted.

```
Recv: //Disk init fail!
Recv: //Disk init fail!
Recv: //############sd_mounted Error! can't open file !
```

#### Hardware / Software Info
PI = Raspberry Pi 4 Model B Rev 1.2p
Printer: FIRMWARE_NAME:CBD made it PROTOCOL_VERSION:V4.3.13_LCDC (Mars2Pro)

#### Description
I checked /etc/fstab and had the entry to piusb.bin twice, maybe that was my fault. I changed it, rebooted, but same. `sudo mount -a` seems to work fine. No error.
Tried 2 different cables and bought even an OTG adapter for the PI (https://www.8086.net/products/#80860015). Same problem. I can move Z Axis (controlled via GPIO) and Octoprint works so far just fine. 

Any good ideas out? 
I would bet I am the layer 8 prob - maybe a hint or a command line command to test/debug the connection would help :)

Apart from that ! Really nice project, really!