rudetrooper/Octoprint-Chituboard #14: Photon Mono SE

etyrnal, 2022-01-24:
How difficult would it be to add something to the plugin which monitored a resin printer's terminal through the Rx/Tx and display it in an area of the plugin's window?  Idea being to make it easier to add more printers and discover their comms.

So, for example, for a printer which is not supported yet, a person could go to a reveal-able window in the chitu plugin, see the sla printer's comms, and run operations on the sla printer's front panel, and see/capture the commands in the plugin's terminal window?

rudetrooper, 2022-02-05:
It would take quite a bit of work to do that. I would have to rewrite alot of the backend to support this and parts of the front end GUI. Octoprints comm layer doesn't yet support this kind of functionality. It might support it in Octoprint 2.0.0 with the new comm layer but that release isn't due for a while.

etyrnal, 2023-02-01:
Is it still unlikely this will ever work on the Photon Mono SE?  I had originally purchased the comb of of the Mono SE and the RPI Zero 2 W just for this purpose.   If it's not likely to ever work, i'll just get rid of the rpi.  Just wondering as i've seen additional activity in the issues section, not for my machine specifically, but just in general.  I made me wonder of anything had been advanced that might just so happen to overlap the Mono SE.  Thanks.

rudetrooper, 2023-02-01:
Unfortunately It doesn't look like I'll be able to get the plugin working with the Anycubic Photon mono SE since I can't get it to accept Gcode commands or any kind of input from the serial terminal. Also Anycubic hasn't been very helpful whenever I attempted to contact them, it seems they locked down the serial communication and refuse to provide any information. I don't exactly have enough time to exhaustively reverse engineer and crack the firmware.

etyrnal, 2023-02-02:
thanks.  that's understandable.