rudetrooper/Octoprint-Chituboard #18: WARNThe received line contains at least one null byte character at position 0, this hints at some data corruption going on

mcdviii, 2022-02-14:
I'm receiving title as an error after selecting a file to print with my Elegoo Mars 2 Pro. 

Latest Octopi is installed on a rPi Zero 2 W.
GRND, Rx & Tx are wired correctly.
I can send commands to the printer that move the Z axis.
I can print uploaded files directly from the printer menu screen.

Console Log:

```
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyS0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x6bfe8330, open=True>(port='/dev/ttyS0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: M4002
Recv: ok N:414
Send: M4002
Changing monitoring state from "Connecting" to "Operational"
Recv: ok N:415
Send: N0 M110 N0*125
Recv: //#
Recv: \x00#checkSum error,i:2 expect:0x7e,actual:0x7d!str:N0*125
WARNThe received line contains at least one null byte character at position 0, this hints at some data corruption going on
Recv: 
Recv: ok
Send: M115
Recv: ok CBD make it.Date:Jul  3 2021 Time:10:42:39
Repl: ok CBD make it.Date:Jul  3 2021 Time:10:42:39
 -> ok FIRMWARE_NAME:CBD made it PROTOCOL_VERSION:V4.13 Date:Jul  3 2021 Time:10:42:39

Send: M21
Recv: ok N:2
Send: M20
Recv: Begin file list
Recv: Vert
Recv: \x00l_Filament_Spool_Parts_Drawer_Box_0.05mm_2h46m.ctb 66843936
WARNThe received line contains at least one null byte character at position 0, this hints at some data corruption going on
Recv: wmetad.metadata.jlament_SpoVertical_Filament_Sm.ctb 44308391
Recv: \x00_Latch_0.05mm_55m.ctb 24740087
WARNThe received line contains at least one null byte character at position 0, this hints at some data corruption going on
Recv: MixerV2_0.05mm_3h23m.ctb 44308391
Recv: End file list
Recv: ok L:4
Send: M4000
Recv: ok N:4
Send: M4000
Recv: ok N:5
Send: M4000
Recv: ok N:6
Send: M4000
Recv: ok N:7
Send: M4000
Recv: ok N:8
Send: M4000
Recv: ok N:9
Send: M4000
Recv: ok N:10
Send: M4000
Recv: ok N:11
Send: M4000
Recv: ok N:12
Send: M4000
Recv: ok N:13
Send: M4000
Recv: ok N:14
Send: M4000
Recv: ok N:15
Send: M4000
Recv: ok N:16
Send: M4000
Recv: ok N:17
Send: M4000
Recv: ok N:18
Send: M4000
Recv: ok N:19
Send: M4000
Recv: ok N:20
Send: M4000
Recv: ok N:21
Send: M4000
Recv: ok N:22
Send: M4000
Recv: ok N:23
Send: M4000
Recv: ok N:24
Send: M4000
Recv: ok N:25
Send: M4000
Recv: ok N:26
Send: M4000
Recv: ok N:27
Send: M4000
Recv: ok N:28
Send: M4000
Recv: ok N:29
Send: M4000
Recv: ok N:30
Send: M4000
Recv: ok N:31
Send: M4000
Recv: ok N:32
Send: M4000
Recv: ok N:33
Send: M4000
Recv: ok N:34
Send: M4000
Recv: ok N:35
Send: M4000
Recv: ok N:36
Send: M4000
Recv: ok N:37
Send: M4000
Recv: ok N:38
Send: M4000
Recv: ok N:39
Send: M4000
Recv: ok N:40
Send: M4000
Recv: ok N:41
Send: M4000
Recv: ok N:42
Send: M4000
Recv: ok N:43
Send: M4000
Recv: ok N:44
Send: M4000
Recv: ok N:45
Send: M4000
Recv: ok N:46
Send: M4000
Recv: ok N:47
Send: M4000
Recv: ok N:48
Send: M4000
Recv: ok N:49
Send: M4000
Recv: ok N:50
Send: M4000
Recv: ok N:51
Send: M4000
Recv: ok N:52
Send: M4000
Recv: ok N:53
Send: M4000
Recv: ok N:54
Send: M4000
Recv: ok N:55
Send: M4000
Recv: ok N:56
Send: M4000
Recv: ok N:57
Send: M4000
Recv: ok N:58
Send: M4000
Recv: ok N:59
Send: M4000
Recv: ok N:60
Send: M4000
Recv: ok N:61
Send: M4000
Recv: ok N:62
Send: M4000
Recv: ok N:63
Send: M4000
Recv: ok N:64
Send: M4000
Recv: ok N:65
Send: M4000
Recv: ok N:66
Send: M4000
Recv: ok N:67
Send: M4000
Recv: ok N:68
Send: M4000
Recv: ok N:69
Send: M4000
Recv: ok N:70
Send: M4000
Recv: ok N:71
Send: M4000
Recv: ok N:72
Send: M4000
Recv: ok N:73
Send: M4000
Recv: ok N:74
Send: M4000
Recv: ok N:75
Send: M4000
Recv: ok N:76
Send: M4000
Recv: ok N:77
Send: M4000
Recv: ok N:78
Send: M4000
Recv: ok N:79
Send: M4000
Recv: ok N:80
Send: M23 Vertical_Filament_Spool_Parts_Drawer_Latch_0.05mm_55m.ctb
Recv: File selected
Recv: \x00ical_Filament_Spool_Parts_Drawer_Latch_0.05mm_55m.ctb Size:24740087
WARNThe received line contains at least one null byte character at position 0, this hints at some data corruption going on
Recv: File selected
Recv: ok N:81
Send: M4000
Recv: ok N:82
Send: M4000
Recv: ok N:83
Send: M4000
Recv: ok N:84
Send: M4000
Recv: ok N:85
Send: M4000
Recv: ok N:86
Send: M4000
Recv: ok N:87
Send: M4000
Recv: ok N:88
Send: M4000
Recv: ok N:89
Send: M4000
Recv: ok N:90
Send: M4000
Recv: ok N:91
Send: M4000
Recv: ok N:92
Send: M4000
Recv: ok N:93
Send: M4000
Recv: ok N:94
Send: M4000
Recv: ok N:95
Send: M4000
Recv: ok N:96
Send: M4000
Recv: ok N:97
Send: M4000
Recv: ok N:98
Send: M4000
Recv: ok N:99
Send: M4000
Recv: ok N:100
Send: M4000
Recv: ok N:101
Send: M4000
Recv: ok N:102
Send: M4000
Recv: ok N:103
Send: M4000
Recv: ok N:104
Send: M4000
Recv: ok N:105
Send: M4000
Recv: ok N:106
Send: M4000
Recv: ok N:107
Send: M4000
Recv: ok N:108
Send: M4000
Recv: ok N:109
Send: M4000
Recv: ok N:110
Send: M4000
Recv: ok N:111
Send: M4000
Recv: ok N:112
Send: M4000
Recv: ok N:113
Send: M4000
Recv: ok N:114
Send: M4000
Recv: ok N:115
Send: M4000
Recv: ok N:116
Send: M4000
Recv: ok N:117
Send: M4000
Recv: ok N:118
Send: M4000
Recv: ok N:119
Send: M4000
Recv: ok N:120
Send: M4000
Recv: ok N:121
Send: M4000
Recv: ok N:122
Send: M4000
Recv: ok N:123
Send: M4000
Recv: ok N:124
Send: M4000
Recv: ok N:125
Send: M4000
Recv: ok N:126
Send: M4000
Recv: ok N:127
Send: M4000
Recv: ok N:128
Send: M4000
Recv: ok N:129
Send: M4000
Recv: ok N:130
Send: M4000
Recv: ok N:131
Send: M4000
Recv: ok N:132
Send: M4000
Recv: ok N:133
Send: M4000
Recv: ok N:134
Send: M4000
Recv: ok N:135
Send: M4000
Recv: ok N:136
Send: M4000
Recv: ok N:137
Send: M4000
Recv: ok N:138
Send: M4000
Recv: ok N:139
Send: M4000
Recv: ok N:140
Send: M4000
Recv: ok N:141
Send: M4000
Recv: ok N:142
Send: M4000
Recv: ok N:143
Send: M4000
Recv: ok N:144
Send: M4000
Recv: ok N:145
Send: M4000
Recv: ok N:146
Send: M4000
Recv: ok N:147
Send: M4000
Recv: ok N:148
Send: M4000
Recv: ok N:149
Send: M4000
Recv: ok N:150
Send: M4000
Recv: ok N:151
Send: M4000
Recv: ok N:152
Send: M4000
Recv: ok N:153
Send: M4000
Recv: ok N:154
Send: M4000
Recv: ok N:155
Send: M4000
Recv: ok N:156
Send: M4000
Recv: ok N:157
Send: M4000
Recv: ok N:158
Send: M4000
Recv: ok N:159
Send: M4000
Recv: ok N:160
Send: M4000
Recv: ok N:161
Send: M4000
Recv: ok N:162
Send: M4000
Recv: ok N:163
Send: M4000
Recv: ok N:164
Send: M4000
Recv: ok N:165
Send: M4000
Recv: ok N:166
Send: M4000
Recv: ok N:167
Send: M4000
```

rudetrooper, 2022-02-14:
Can you send provide the sliced file and the firmware version so I can try and replicate the issue? Are you starting the prints by selecting the files within the resin folder in the file menu? Also can you provide me the .octoprint/config.yaml file from your pi, the serial settings might be useful.

mcdviii, 2022-02-14:
Absolutely,

This was downloaded directly from the /resin directory:

https://www.dropbox.com/s/bdownge4ti8mpm3/MixerV2_0.05mm_3h23m.ctb?dl=0

**~/.octoprint/config.yaml**:

https://www.toptal.com/developers/hastebin/ecodubovab.yaml

Firmware ver EL3D-3.0.2

The behavior does not change even when I select a file to print under the /resin folder, but I tried to favor those files after I found that's the directory I needed to be using.

rudetrooper, 2022-02-17:
Thanks, I'll look into it and try to replicate the issue.

mcdviii, 2022-02-17:
Thanks for your time & the update on progress. I took another look at my wiring last night & saw no issues. The Pi is pulling under 500mA and at no point goes over that. I'll look into the buck converter, but I feel like the issue is either a misconfiguration or software bug for the moment.

mcdviii, 2022-02-20:
I've looked into this more; it may still be a wiring issue. My custom USB harness only includes the two data cables & no reference to ground. I will undo my USB wiring & report back--don't wrack your brain too hard on this until I do.

mcdviii, 2022-02-20:
I've added Ground to the USB data cable with no change in behavior. Which option should I use to upland files with, the Upload button or Upload to SD? Using Upload under /resin works as expected. Upload to SD throws an error about invalid filetype.

rudetrooper, 2022-02-22:
Upload to SD doesn't work. becuase I overrided that method when I wrote the plugin. Just use the regular upload button, also I tested the file you sent and it works fine on my Elegoo Mars 2 and personal Octoprint setup.

mcdviii, 2022-02-22:
Any ideas on what to troubleshoot next? Did I miss any setup steps?

Thanks

mcdviii, 2022-04-19:
I've been making due by selecting prints from the screen. It has been working well until now. Now I'm getting failed prints where the machine stalls indefinitely on random layers & appears to be unable to read from the file. Similar issues happened with the included USB stick, but the machine recovered with a short "Memory error" message which disappeared once the machine recoverd. Wondering if it's still a hardware interface issue, my control board, or a problem with the formatting of the SAMBA partition.

I've switched SD cards in the Pi with no improvement. If anyone is aware of a better firmware version I could be using, please advise. Thanks!

xmodpt, 2022-04-22:
What i would try in your case is unplug the pi fro the usb and power. try printing with a normal USB Pen drive if no issues, 2th step would be change the usb cable from pi's data port to front of printer for a new one with the pads correctli taped as per reference.

CrAzY RaBbIt

mcdviii, 2022-04-22:
I'm confident of the printer end of the data cable being configured correctly. I'm looking at the Pi end currently. I'm also looking a 3rd time for cold joints. Will report back as time allows.

xmodpt, 2022-04-22:
one VERY DUM question... what type of power are you using? it is knowed that the PI is very "Special" in regards to power

mcdviii, 2022-04-22:
I'm using an adjustable DC/DC buck converter, pulling from the line-in controller board input terminals. It's set to 5v.

mcdviii, 2022-09-13:
I finally had time to approach this issue again.

I've confirmed it isn't a data wiring issue by using the tape method on a data USB cable I had laying around. My custom USB cable provides the same results.

I may try to find a shielded cable to see if that improves print stalls.

I'm still leaning toward the original issue being a misconfiguration or firmware bug.

mnbf9rca, 2022-09-13:
i'm also seeing this problem. i'm confident the wiring is fine. Printing from direct USB works fine.

mcdviii, 2022-09-13:
Upon closer inspection, I'm running firmware very. 4.4.3. I was mistakenly reading what I think is the Chituboard version as the firmware version. Mariner software has documented issues with 4.4.3. I will downgrade to older firmware from the Elegoo firmware page for Mars 2 Pro & report back with results.

mcdviii, 2022-09-14:
I've downgraded my printer firmware for my Mars 2 Pro from V4.4.3-F9.23 to V4.3.13-F2.23 (using the installation steps outlined on the official firmware page.

The results are that I am able to print from the Octoprint dashboard web interface, as expected (albeit, with a delay/lag). The original issue is solved.

For more info on why 4.4.3 is problematic, see:

https://github.com/luizribeiro/mariner/issues/453

Thanks for everyone's time.

closed 2022-09-14 by mcdviii

mnbf9rca, 2022-09-17:
Unfortunately there’s no earlier version for the Jupiter :(

mcdviii, 2022-09-17:
> Unfortunately there’s no earlier version for the Jupiter :(

Consider using the Mariner fork referenced in the issue I linked to above:

https://github.com/BlueFinBima/mariner/releases/tag/v0.2.0-1

Keep in mind--the fork is abandonware & not officially supported by any project. It may be that hack works for you.

mnbf9rca, 2022-09-20:
Thanks for the pointer. i had to do some work to update it and repackage it so it would install, which i've now done and the package is available at https://github.com/mnbf9rca/mariner/releases/tag/0.2.0-2