rudetrooper/Octoprint-Chituboard #22: using a single pi for Mars and Ender

tusharpc, 2022-03-29:
hi, a question, I have both the Mars Pro 2 and an Ender 3 pro side by side. I'm currently using octoprint successfully on the ender via a pi Z2W. can i run both? aka have both printers connected to usb hub that is connected to the pi. if you install the chitu plugin, the controls for the FDM printer are lost, is there any way to revert that or disable to plugin to just run the ender 3pro or vice versa ?. the whole point is to use a single pi z2w, given the shortage.

by the way, this is fantastic work !

rudetrooper, 2022-04-12:
Its not recommended to use a single pi to control an FDM and resin printer. This application is emulating a flash drive using the microSD and the USB OTG port so adding extra traffic that could interfere with the CPU emulating the flash device could cause significant problems or print failures.

tusharpc, 2022-04-12:
Thanks, that make sense, my question was really about using one or another not both simultaneously.
So once I install your plugin, is there a way to temporarily disable it's functionality so I can now use the octoprint default ui for ender 3

rudetrooper, 2022-04-13:
You can disable the plugin via the Octoprint plugin manager in the settings menu. That has always been possible.

closed 2022-05-02 by rudetrooper