rudetrooper/Octoprint-Chituboard #25: octoprint-chituboard unresponsive to printing a compatible file

madelbrot, 2022-06-11:
Hey! This is a partial re-post from the [https://community.octoprint.org/t/octoprint-chituboard-software-unresponsive-to-printing-a-compatible-file/44920](octoprint forum). please see that link for relevant downloadable files. 

First of all, thank you to rudetrooper and everyone in the community! I am really thankful for everyone's help and good will.

**What is the problem?**
when I press "print" on a compatible file, absolutely nothing happens

**What did you already try to solve it?**
_firmware:_

changed the firmware (asked Elegoo to send over an octoprint compatible set, please see link below)

[https://community.octoprint.org/t/mars-firmware-for-chituboard-compatibility/44919](link to firmware post)

_software_:

various debugging procedures such as :
trying files that I knew would fail (interestingly those have a different outcome, see [logs](https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299). "_20x20x20.pw0" was a file I knew would fail and you can see in the log that it does throw an error message)
printing the target file directly from USB (works perfectly)
checking that I can manipulate the z axis from octopi (I can, and it always feels so cool lol)
searching for every single suspicious thing online (disk init fail searched, searched the warning about null byte characters, searched about the odd outputs of M110 checkSum error, M21 disk, checked the github, this forum, everywhere!)
used chitubox basic and chitubox pro of multiple versions to slice example files and send them over. these files all print fine on the printer when direct printing from usb stick, but do not send over octoprint

_hardware:_

opened up my USBA <> USBA cable to make sure there are data lines and shielding. all are present. Tape is affixed over the 5V pin.
checked the motherboard for observable issues
my guess is it is something about the communication of the file. out of my depth here, but I think because I get so many Disk Init errors, there may be something about how the pi sends / printer receives data that is in disagreement. but I have been working on this for just over a week and haven't come up with an answer! seems like people may be having similar problems as there are some related topics in the issues section of the octoprint-chituboard github that remain unresolved.

### bottom line 

Apparently there is a problem with the chituboard plugin that may be causing the issue:

`  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1508, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/files.py", line 865, in gcodeFileCommand
    printer.select_file(filenameToSelect, sd, printAfterLoading, user)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_chituboard/sla_printer.py", line 130, in select_file
    sliced_model_file = file_format.read(Path(path_on_disk))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_chituboard/file_formats/ctb.py", line 206, in read
    ctb_param = CTBParam.unpack(file.read(CTBParam.get_size()))
  File "/home/pi/oprint/lib/python3.7/site-packages/typedstruct.py", line 89, in unpack
    return cls(*unpack(cls.get_format(), buffer))
struct.error: unpack requires a buffer of 60 bytes`

Could you please help me navigate this issue? I am happy to help in whatever capacity I can. This is pretty far out of my depth but I am happy to run tests and try different troubleshooting attempts.

rudetrooper, 2022-06-15:
I'll work on fixing the issue. Can you send me the file that caused the original issue? Also what version of CTB file was it and which slicer did you use to generate it?

marceloSchardong, 2022-09-29:
Hi everyone, I'm having the same problem here.
I'm currently running a custom printer on ChiTu E10 Lite mainboard, 12.8 inch 6k Monochrome LCD. I'm connected to the printer(via Raspberry pi 4) and have control of the Z motor, but hein I try to print something, nothing happens. 

The slicer I'm using is Chitubox, the files are .ctb. I'm able to send the files to the Raspberry pi, but nothing happens when I try to print(no even on terminal).

I'll be glad to help on any test print or try run.