rudetrooper/Octoprint-Chituboard #26: Suddenly not connecting via serial, also the serial port stays open after failure.

MythikWolf, 2022-06-29:
This was working.  Suddenly not working.

Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyS0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0xac54abf0, open=True>(port='/dev/ttyS0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: M4002
No answer from the printer within the connection timeout, trying another hello
Send: M4002
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyS0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Unexpected error while connecting to serial port /dev/ttyS0, baudrate 115200 from hook default: error: '(22, 'Invalid argument')' @ comm.py:_open_serial:3836

Seems it stays stuck afterwards.  Working with an original Anycubic Photon.  As mentioned it was working fine.  None of the connections look like they got knocked loose.

MythikWolf, 2022-06-29:
Nvm.  It was just a dying Pi.  I'll give it a good burial.

rudetrooper, 2022-06-30:
RIP Pi

rudetrooper, 2022-06-30:
Also It might just be the microSD card that died not the Pi.

MythikWolf, 2022-06-30:
Actually transferred it to a new Pi. Works fine. RIP Pi.

closed 2022-07-03 by MythikWolf