rudetrooper/Octoprint-Chituboard #28: [Request] Support latest firmwares

KairuByte, 2022-08-25:
I am fully aware there is a standing "I will not support their new terrible encryption" response, but I'd like to make an official request to support it.

There are some good arguments *against* support, the main one being not wanting to support and further their terrible walled garden approach. I absolutely understand this stance, and to a certain point agree with it. Though I would argue that showing the entire endeavor to be pointless by reverse engineering the encryption would be a better way to make that statement, as they are essentially being rewarded for their ill intent with exactly what they wanted.

There are some, in my opinion, bad arguments *against* support, the main one being that it would somehow get OctoPrint into legal hot water. Unless this plugin is directly tied to OctoPrint, there's really no danger of it getting into trouble. That would be akin to M$ getting sued because you wrote a Visual Studio plugin that facilitates a crime. I'd also point out, there's not much chance they would be able to sue or take down this plugin, assuming you don't release any copyrighted materials.

In contrast, again in my opinion, there are three main arguments *for* support.

The first, is that older firmware is hard to come by, and harder to get loaded. I've been strugging to get my LD-002R onto older firmware, and I'm not even certain I have the proper firmware files.

The second, there seem to be printers released with no path to downgrade. My understanding is that there are some printers that have no proper path to the older firmware? There is also a high probability that future versions will be locked to the newer firmware, since they are attempting to lock you into their slicer.

The third, is that it is highly likely any firmware moving forward will have this, or similar, encryption in it. This means that if any bug or oddity is ever patched, end users will essentially have a choice between using OctoPrint, or patching a problem.

rudetrooper, 2022-12-25:
I'm working on adding support for the CTBv4 files in a way that doesn't mess up support for the for prior file version. Also it seems like Chitu might have changed the gcode for the newer versions of the firmware so I'm working on making sure that adding support for newer firmware doesn't impact support for older more functional versions. Though the lack of a firmware downgrade path for printers  is something users should bring up with Chitu, not me. Personally I prefer to work with open source firmware that has more functionality and customization options.

KairuByte, 2022-12-25:
I agree with you on all fronts, including the main place to make the point about downgrade paths being Chitu. However my two attempts to make contact with them were met with silence. That said, I was mainly just pointing it out as a roadblock to using the plugin.