rudetrooper/Octoprint-Chituboard #32: the chithuboard plugin didn't work for Elegoo Satrun2,  cannot  print from ctb files in the SD card

biolifei, 2022-12-17:
My Elegoo Saturn2 printer cannot start to print any uploaded files on the 64 GB SD card when it is operational after connected to Octoprint server.
 I installed OctoPi version: 0.18.0 with OctoPrint version: 1.8.6 by using the Raspberry Pi 4B connected through /dev/ttyS0 baudrate: 115200. The octopi is installed with octoprint plugins chituboard.
 The priner is connected by the jumper wires from the Raspberry Pi’s GPIO pins to the Elegoo Saturn2 motherboard. The TX pin is connected to the motherboard’s RX pin, GND to GND, Rx to Tx.
 The Pi USB connector is connected to the “USB Disk” USB port on the printer’s motherboard with the USB Power Blocker to strip 5V from a USB cable.
2022-12-17 16:42:47,284 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:132.199.73.186
2022-12-17 16:42:47,322 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2022-12-17 16:42:47,566 - octoprint.filemanager.analysis - INFO - Analysis of entry local:resin/4W5T_ribbons_Hbonds.ctb finished, needed 6.81s
2022-12-17 16:42:47,832 - octoprint.server.util.flask - INFO - Passively logging in user saturn2 from ::ffff:132.199.73.186
2022-12-17 16:42:47,833 - octoprint.access.users - INFO - Logged in user: saturn2
2022-12-17 16:42:48,525 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python3 -m pip" as command to invoke pip
2022-12-17 16:42:48,531 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python3.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2022-12-17 16:42:48,532 - octoprint.util.pip - INFO - ==> pip ok -> yes
2022-12-17 16:42:48,569 - octoprint.plugins.softwareupdate - INFO - Minimum free storage across all update relevant locations is 48.3GB. That is considered sufficient for updating.
2022-12-17 16:42:48,569 - octoprint.plugins.softwareupdate - INFO - Not fetching check overlays, we are offline
2022-12-17 16:42:48,605 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Totoro2.ctb
2022-12-17 16:42:49,261 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2022-12-17 16:42:49,461 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2022-12-17 16:43:05,029 - octoprint.plugins.bedlevelvisualizer - INFO - OctoPrint-BedLevelVisualizer loaded!
2022-12-17 16:43:05,030 - octoprint.plugins.chituboard - INFO - Octoprint-Chituboard plugin startup
2022-12-17 16:43:05,539 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 13, 'printer_state': 'OPERATIONAL'}
2022-12-17 16:43:06,287 - octoprint.plugins.tracking - INFO - Sent tracking event pong, payload: {'version': '1.8.6', 'os': 'linux', 'bits': 32, 'python': '3.7.3', 'pip': '20.3.3', 'cores': 4, 'freq': 1500.0, 'ram': 3971166208, 'pi_model': 'Raspberry Pi 4 Model B Rev 1.2', 'octopi_version': '0.18.0', 'octopiuptodate_build': '0.18.0-1.8.6-20221018093204', 'plugins': 'printtimegenius:2.3.0,fixcbdfirmware:0.4.0,chituboard:0.1.4,octolapse:0.4.1,bedlevelvisualizer:1.1.1,abl_expert:0.6'}
2022-12-17 16:43:06,484 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'base_url': 'http://132.199.74.255/', 'path': '/', 'query_string': 'l10n=en'}
2022-12-17 16:43:06,625 - octoprint.plugins.tracking - INFO - Sent tracking event startup, payload: {'version': '1.8.6', 'os': 'linux', 'bits': 32, 'python': '3.7.3', 'pip': '20.3.3', 'cores': 4, 'freq': 1500.0, 'ram': 3971166208, 'pi_model': 'Raspberry Pi 4 Model B Rev 1.2', 'octopi_version': '0.18.0', 'octopiuptodate_build': '0.18.0-1.8.6-20221018093204'}
2022-12-17 16:43:13,018 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2022-12-17 16:43:13,833 - octoprint.filemanager.analysis - INFO - Analysis of entry local:Totoro2.ctb finished, needed 10.02s
2022-12-17 16:43:13,986 - octoprint.server.util.sockjs - INFO - User saturn2 logged in on the socket from client ::ffff:132.199.73.186
2022-12-17 16:43:14,218 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2022-12-17 16:43:14,837 - octoprint.filemanager.analysis - INFO - Starting analysis of local:little_goku.ctb
2022-12-17 16:43:21,570 - octoprint.filemanager.analysis - INFO - Analysis of entry local:little_goku.ctb finished, needed 6.73s
2022-12-17 16:43:22,608 - octoprint.filemanager.analysis - INFO - Starting analysis of local:little_goku_6h.ctb
2022-12-17 16:43:30,367 - octoprint.filemanager.analysis - INFO - Analysis of entry local:little_goku_6h.ctb finished, needed 7.76s
2022-12-17 16:43:31,975 - octoprint.util.jinja - WARNING - Loading plugin template '/webcams/mjpg_streamer/logitech_c920.jinja2' from '/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/templates/webcams/mjpg_streamer/logitech_c920.jinja2' without plugin prefix, this is deprecated and will soon no longer be supported.
2022-12-17 16:43:31,987 - octoprint.util.jinja - WARNING - Loading plugin template '/webcams/mjpg_streamer/raspi_cam_v2.jinja2' from '/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/templates/webcams/mjpg_streamer/raspi_cam_v2.jinja2' without plugin prefix, this is deprecated and will soon no longer be supported.
2022-12-17 16:43:32,002 - octoprint.util.jinja - WARNING - Loading plugin template '/webcams/mjpg_streamer/logitech_c250.jinja2' from '/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/templates/webcams/mjpg_streamer/logitech_c250.jinja2' without plugin prefix, this is deprecated and will soon no longer be supported.
2022-12-17 16:43:34,911 - octoprint.server.preemptive_cache - INFO - ... done in 28.43s
2022-12-17 16:43:44,900 - octoprint.util.comm - INFO - Communication timeout while idle, trying to trigger response from printer.
2022-12-17 16:44:27,459 - octoprint.server.util.flask - INFO - Passively logging in user saturn2 from ::ffff:132.199.73.186
2022-12-17 16:44:27,460 - octoprint.access.users - INFO - Logged in user: saturn2
2022-12-17 16:44:27,739 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:132.199.73.186
2022-12-17 16:44:28,804 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:132.199.73.186
2022-12-17 16:44:28,852 - octoprint.server.util.flask - INFO - Passively logging in user saturn2 from ::ffff:132.199.73.186
2022-12-17 16:44:28,852 - octoprint.access.users - INFO - Logged in user: saturn2
2022-12-17 16:44:30,113 - octoprint.server.util.sockjs - INFO - User saturn2 logged in on the socket from client ::ffff:132.199.73.186
2022-12-17 16:44:30,587 - octoprint.plugins.tracking - INFO - Sent tracking event webui_load, payload: {'browser_name': 'Chrome', 'browser_version': '108.0.0.0', 'os_name': 'Windows', 'os_version': '10'}
2022-12-17 16:44:35,739 - octoprint - ERROR - Exception on /api/files/local/little_goku.ctb [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_chituboard/sla_printer.py", line 126, in select_file
    sliced_model_file = file_format.read_dict(Path(path_on_disk),fileData["analysis"])
KeyError: 'analysis'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1519, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1517, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1503, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1598, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/files.py", line 874, in gcodeFileCommand
    printer.select_file(filenameToSelect, sd, printAfterLoading, user)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_chituboard/sla_printer.py", line 130, in select_file
    sliced_model_file = file_format.read(Path(path_on_disk))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_chituboard/file_formats/ctb.py", line 206, in read
    ctb_param = CTBParam.unpack(file.read(CTBParam.get_size()))
  File "/home/pi/oprint/lib/python3.7/site-packages/typedstruct.py", line 89, in unpack
    return cls(*unpack(cls.get_format(), buffer))
struct.error: unpack requires a buffer of 60 bytes_

biolifei, 2022-12-17:
[image]

rudetrooper, 2022-12-25:
The plugin doesn't yet support the CTBv4 files, I'll fix it once I have some time.

DonChooluvit, 2023-01-11:
That explains it. I just finished setting everything up and was able to get the print bed to home and move, but the files were not visible.

avoyles, 2023-06-06:
> The plugin doesn't yet support the CTBv4 files, I'll fix it once I have some time.

Any updates on this? I've got a Saturn2 and would desperately love to have this working. If there's anything I can do to assist, I'm more than happy to test/help.

Spyder19, 2023-08-27:
I'm having an odd problem that sounds related.

Files that I slice with chitubox aren't accepted by the printer. I assume that's because I didn't update the firmware. However, files I slice in lychee work just fine. 

Then I added the pi02w, and now it's telling "invalid file type" when I try to upload. 

I wonder, if I slice it in chitu, would the file transfer ? And if it does, would the printer accept it without the firmware update which would preclude me from using the CTBv3 files anyway

The error says that it only wants gcode, but it doesn't say that til it's almost 100% uploaded. I have to drag n drop. It doesn't like file transfer thru octoprint file select