rudetrooper/Octoprint-Chituboard #37: Photon M3 Support?

BzowK, 2023-02-22:
Hey Guys - 

Just curious if there's Photon M3 support.  Was going to test so disassembled it this evening, but pins aren't labeled like they are on other boards so not sure how to proceed as obviously don't want to break printer.  Below is a pic of the motherboard.  Headers which were populated / connected are highlighted in red.  The bottom of the board had no silkscreen.  Any thoughts?  Thanks!

[image]

rudetrooper, 2023-02-24:
Unfortunately it doesn't support it, I don't know what instructions the anycubic printers take via the UART. I just know it's not gcode like the Chitu based printers.

BzowK, 2023-02-25:
Disappointing - Thanks

closed 2023-02-25 by BzowK

TheFloydman, 2023-03-13:
This is disappointing to find after I've already done all the setup for my Photon M3 Max. The board has Rx and Tx pins labeled. Is there anything I could do to help in this situation? If possible, I'd request that a note be added to supported printers that Photon M3 is _not_ supported. Including Photon in that list and stating that Anycubic boards are supported gives the impression that M3 is supported. (Not a criticism. Just a helpful edit for others who come after and don't search closed issues.)