rudetrooper/Octoprint-Chituboard #40: planning an update?

linux-paul, 2023-03-03:
Will there be an update for ctbV4 and ctbV4_encrypted support soon? I have started working on an implementation of ctbV4_encrypted myself. :)

rudetrooper, 2023-03-08:
Yeah, I have it mostly finished. Just need to finish some final bits before release, some life events got in the way. Should be released with a week or 2.

linux-paul, 2023-03-09:
Pretty cool, Thank You, :)

DonChooluvit, 2023-03-09:
That is AWESOME!! Looking forward to it. Again, I will happily test on a Saturn 2

Millerboy3, 2023-03-14:
> Yeah, I have it mostly finished. Just need to finish some final bits before release, some life events got in the way. Should be released with a week or 2.

thats amazing news!

bvaerewyck, 2023-03-16:
You are a rockstar!  I have two orange pi pc's with the OTG ready.   I just need to get the ventilation system ready.

SnowFarmsEngineering, 2023-03-16:
Great! I am looking forward to seeing if I can get my Mars 3 working with your plugin. Thank you!

mbudnek, 2023-08-06:
@rudetrooper Any update on this?  I'd love to start using this plugin with my Saturn 2.

ChristophCaina, 2023-10-12:
Hi :)
Would be really nice to know, if there's an update about the new ctbv4 format :)
I was just ready to start with Octoprint for my Mars2Pro - but then I've heard, that the new ctb4 are not supported...