rudetrooper/Octoprint-Chituboard #42: Chitu M21 serial connection

jpg0, 2023-03-20:
I have an Elegoo Mars 3 Pro which I have recently attempted to connected to octoprint. It is a newer ctbv4 firmware printer, so I am not expecting encrypted file support. I did however expect to be able to get serial communication working, but have failed to do so. The board is a Chitu M21, which doesn't seem to be used elsewhere. It does have plainly marked pins, however I have not been able to get any serial communication going, after connecting the GND, TX, and RX pins as in the photo.

Any suggestions? 

Thanks

[image]

jpg0, 2023-03-27:
I also tried on both a pi and a macbook, also with varying the baud rate. I'm fairly sure the voltage is correct as the rx and tx pins are both at 3.3V, but there is no comms happening. I suspect that Chitu have disabled the serial communication.

bvaerewyck, 2023-03-27:
Did you reverse the tx and rx?  I mad that mistake a few years ago.  Sorry if I missed that and you already tried.  Hope it works.

jpg0, 2023-03-27:
Yeah, I tried both ways, lots of times. Pretty sure there is nothing coming out.

bvaerewyck, 2023-03-27:
Is it the adafruit cable?  Did you test continuity?  Bad cable?

jpg0, 2023-03-28:
I used some random cable that I had, although I did test continuity with a multimeter beforehand and also tried more than one cable (including a different cable attached to a usb ftdi adapter connected to a laptop).

One of the reasons that I posted here was to see if others saw the same behaviour as me to confirm whether I had made a mistake or there is no comms. So far I am aware of 2 (myself included) people who see no signal, and 0 who see a signal!

I suspect though that we will see more people try this though, as the Mars 3 Pro is fairly new and seems like a pretty good piece of hardware.

linux-paul, 2023-03-28:
I've also teted differtent cables with no effect, Also I tested boudrates up to 1000000.
but ... depending to the fact of the 5VVcc pin and the chip has been changed from ARM controller to FPGA ... maybe a TTL converter will be helpfull in thi case?

jpg0, 2023-03-29:
Well I did check the voltage on the TX and RX pins with a multimeter and they were 3.3V. I can certainly try listening at 5V, although wouldn't you have expected to see _something_ also if listening at 3.3V?

bvaerewyck, 2023-03-29:
I have 9 printers and I always use this cable...  https://www.amazon.com/ADAFRUIT-Industries-954-Serial-Raspberry/dp/B00DJUHGHI/ref=sr_1_1?crid=JCJOJDDYODR7&keywords=adafruit+ttl+cable&qid=1680056620&s=industrial&sprefix=adafruit+ttl+cable%2Cindustrial%2C148&sr=1-1  works without any issues.

linux-paul, 2023-03-29:
@bvaerewyck If you are running a chituboard serial over usb, which pi board are in use? My A3+ just has one usb port, which is configured as dwc usb drive gadget after installation. But driving the printerboard serial with TTL signals meets my expectation and maybe I'll receive my TTL bridges tomorrow.

linux-paul, 2023-03-31:
levelshifter does also not help and .... btw .... the ARM chip has been placed on the backside ...

jpg0, 2023-04-05:
@linux-paul there were no other connection points on the backside were there? I never pulled out my board to check the other side.

linux-paul, 2023-04-05:
I didn't pulled it out completly, just so far to get a quick view for looking to adidional connectors. There are a pretty bunch of chips solderd on the backside, It looks like completely soldered except one chip close to the pinheader. There are no further connectors, so far I think.

zackfuchtel, 2023-07-07:
> I also tried on both a pi and a macbook, also with varying the baud rate. I'm fairly sure the voltage is correct as the rx and tx pins are both at 3.3V, but there is no comms happening. I suspect that Chitu have disabled the serial communication.

I also think it was disabled. Not a single byte to receive.