rudetrooper/Octoprint-Chituboard #43: Unable to Load and Print Files

mookeyj79, 2023-04-26:
Hello! I am having an issue with the plugin where I cannot load and print files. 
I started with a fresh OctoPrint install, ran the setup script, soldered the serial connection onto the motherboard, and setup the USB.

Here's what's working:
- Serial connection is able to connect.
- Homing and moving the build plate
- Printer is able to see the emulated USB drive
- I am able to see the file details of the file I want to print

Here's what's not working:
- The load button
- The load & print button

Here's my setup:
- Anycubic Photon with the ChiTu L V2 board upgrade
- Raspberry Pi Zero W / Orange Pi Zero 2 (I tried both with the same results.)
- .CTB file slice in both ChituBox & Lychee

Attached our my Octoprint logs. Thank you for your help, and please let me know if you need anymore information!
[octoprint-logs.zip](https://github.com/rudetrooper/Octoprint-Chituboard/files/11328333/octoprint-logs.zip)

doucettom, 2023-05-26:
Make sure you are not slicing a CTBv4 file.

grimsdagger, 2023-05-29:
I have the same error on a phrozen sonic mighty 4k, non updated firmware, and I verified it was a ctbv3 file. I can also see the file on the printer, which printed fine. Opening and printing does not work on octoprint.
This is what the log says:
2023-05-29 11:23:54,254 - octoprint - ERROR - Exception on /api/files/local/resin/raketovy motor V2 (2)_0.05mm_4h2m.ctb [POST]
Traceback (most recent call last):
  File "/home/m/oprint/lib/python3.9/site-packages/octoprint_chituboard/sla_printer.py", line 126, in select_file
    sliced_model_file = file_format.read_dict(Path(path_on_disk),fileData["analysis"])
KeyError: 'analysis'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/m/oprint/lib/python3.9/site-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/m/oprint/lib/python3.9/site-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/m/oprint/lib/python3.9/site-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/m/oprint/lib/python3.9/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/m/oprint/lib/python3.9/site-packages/octoprint/server/util/flask.py", line 1613, in decorated_view
    return func(*args, **kwargs)
  File "/home/m/oprint/lib/python3.9/site-packages/octoprint/server/api/files.py", line 874, in gcodeFileCommand
    printer.select_file(filenameToSelect, sd, printAfterLoading, user)
  File "/home/m/oprint/lib/python3.9/site-packages/octoprint_chituboard/sla_printer.py", line 130, in select_file
    sliced_model_file = file_format.read(Path(path_on_disk))
  File "/home/m/oprint/lib/python3.9/site-packages/octoprint_chituboard/file_formats/ctb.py", line 212, in read
    printer_name = file.read(ctb_slicer.machine_size).decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xcd in position 0: invalid continuation byte

mookeyj79, 2023-06-03:
So i finally got this to work by upgrading the motherboard from the ChiTu L V2 board to the ChiTu L V3 board. Using lychee, it does not matter if a select v2, v3, or v4 ctb, they all work with the plugin.

grimsdagger, 2023-06-03:
After formatting the pi, and reinstalling octoprint, the plugin is also working for me

ritik-kansal, 2023-09-28:
does it work with Phrozen sonic mighty 8k?