rudetrooper/Octoprint-Chituboard #44: Install script defaults to the pi user which may not exist

protoz, 2023-05-03:
Last year, the Raspberry Pi Foundation no longer ships with the default user `pi` this has caused my pi to fail to boot after restarting when it fails to mount the disk that does not exist.

It would be nice to have a prompt to ask which user to install with and default to pi if the user does not input anything.  The first reboot with the password change should only happen if you actually change the password.  I end up commenting out that first block because it's not very useful.

I can open a branch to make these changes if that helps

rudetrooper, 2023-05-03:
Opening a branch would be helpful

protoz, 2023-05-10:
Added a PR: #[46](https://github.com/rudetrooper/Octoprint-Chituboard/pull/46)

rudetrooper, 2023-05-20:
It looks good, PR accepted

closed 2023-05-20 by rudetrooper