rudetrooper/Octoprint-Chituboard #47: Plugin does not work with Octoprint 1.90

DonChooluvit, 2023-05-25:
After updating to the newest version of Octoprint, the webcam does not load, and the controls are just text. I had to uninstall the plugin to regain camera monitoring (which on my Saturn 2, is all I can do as it is).
[image]
[image]

Millerboy3, 2023-05-25:
How do you have your camera setup on the Saturn 2?  I was trying to do the same on my Mars 2 pro

Millerboy3, 2023-05-25:
I also have octoprint 1.9 and don't have that issue.  I'm wondering if it is because of the way I have my camera setup.  I used kiauh initially to set mine up.

DonChooluvit, 2023-05-25:
> How do you have your camera setup on the Saturn 2? I was trying to do the same on my Mars 2 pro

I'm using the IR Pi-cam mounted on the outside. See's right through the case with no issues.
[image]