rudetrooper/Octoprint-Chituboard #49: Sla-control tab broken on latest Octoprint

jeceka201, 2023-08-04:
Just did a fresh install and found the Sla-control tab broken when I enable the webcam. The webcam itself is up and running (I checked the stream url.)

When I disable the cam in the settings, the tab and the controls for my printer are working. 

When I disable the chituboard plugin, the cam is working just fine. 

It might have to do with Octoprint changing the way it handles camera's now (and in effect killing the chituboard_tab.jinja2 template) but my coding skills to confirm this are lacking to confirm.

Octopi: Build 2023.07.20.144556, based on OctoPi 1.0.0, running on Raspberry Pi Zero 2 W Rev 1.0
Octoprint version: 1.9.2 (new camera stack)

[image]

doucettom, 2023-08-22:
I'm also having this bug.

xmodpt, 2023-12-31:
it has nothing to do with the plugin, it's an issue with octoprint it self as i have the same issue but with a FDM printer.
refer to this video for a workarround

https://www.youtube.com/watch?v=L3Sr8p9RdMw&ab_channel=JerryTravisSmith

regards
NP

xmodpt, 2024-01-03:
check my post

[](https://github.com/rudetrooper/Octoprint-Chituboard/issues/51)